iRobotEducation/create3_sim #130: WallFollow uses max_runtime Coverage Server Example uses max_duration

slowrunner, 2021-11-06:
It was a little confusing to find the action goal parameter max_runtime for the new wall_follow action, after using max_duration with the coverage server example.

```
ros2 action send_goal /wall_follow irobot_create_msgs/action/WallFollow "{follow_side: 1, max_runtime:{sec: 1000,nanosec: 0}}"

ros2 action send_goal /coverage create3_examples_msgs/action/Coverage "{explore_duration:{sec: 500, nanosec: 0}, max_duration:{sec: 1000,nanosec: 0}}"

```

justinIRBT, 2021-11-08:
@alsora Should we be consistent between the two?  It would be easier to change the example goal.

alsora, 2021-11-08:
Yes, I agree on changing the example.

slowrunner, 2021-11-08:
Ok - I'll open it on the example.  Thanks.

closed 2021-11-08 by slowrunner