iRobotEducation/create3_sim #135: Use reliable QoS for publishers

eborghi10, 2021-12-06:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

rjcausarano, 2021-12-06:
@eborghi10 Do you know of any issues that this change will fix? Do we run the risk of introducing new issues?

rjcausarano, 2021-12-06:
This might fix https://github.com/iRobotEducation/create3_sim/issues/125

eborghi10, 2021-12-07:
> @eborghi10 Do you know of any issues that this change will fix? Do we run the risk of introducing new issues?

On the opposite, I think it'll allow us to use RViz and another tooling.

eborghi10, 2021-12-07:
> This might fix #125

@Ednip7 Let's double-check this issue too once you have the changes.

closed 2022-03-03 by alsora