iRobotEducation/create3_sim #157: Use `angles` library for math operations

alsora, 2022-02-24:
**Is your feature request related to a problem? Please describe.**
There are some operations on angles in the `irobot_create_toolbox` package.
Rather than implementing them here, we should rely on the `angles` library that is already a dependency of the project.

ckavanagh21, 2023-07-03:
Can you be more specific about where the math regarding angles is? I want to help if I can.

ckavanagh21, 2023-07-07:
[This](https://github.com/ckavanagh21/create3_sim/blob/main/irobot_create_common/irobot_create_toolbox/include/irobot_create_toolbox/math.hpp) is what I was able to come up with. I don't know how to run C++ files, so I couldn't test it. I just looked up the syntax for the `angles` library and modified the code. Do you think this looks right? Also, does the polar_coordinates.hpp file have to be changed as well?

ksatyaki, 2024-11-28:
This might be a different issue, but somewhat related. 

[Here](https://github.com/iRobotEducation/create3_sim/blob/6a8bc44f9be2c260bc17e753c31455d6759af238/irobot_create_ignition/irobot_create_ignition_toolbox/include/irobot_create_ignition_toolbox/utils.hpp#L39C1-L63C2) The calculations for transforming a pose are unnecessarily complicated. I am not even sure the 2D transformation equations are correct! Why not simply use `tf2::Transform * tf2::Transform` (just multiply the transformations). We need (world to some static link). That is simply (world to floating link) * (floating link to static link). Right?