iRobotEducation/create3_sim #170: Bumper sometimes is offset from body in Rviz

roni-kreinin, 2022-03-23:
**Describe the bug**
If you launch `robot_description.launch.py` followed by Rviz2 sometimes the bumper will appear offset from the body in Rviz.
Collisions have the same effect. Restarting Rviz fixes the issue.

If Rviz is launched before the description, the model will spawn normally. Only seems to occur if the description is launched first, although it is not consistent.

**To Reproduce**
Steps to reproduce the behavior:

`ros2 launch irobot_create_common_bringup robot_description.launch.py`

`ros2 launch irobot_create_common_bringup rviz2.launch.py`

**Expected behavior**
Bumper is correctly attached to body.

**Screenshots**
[image]

[image]

**Additional context**
Add any other context about the problem here.

gbr1, 2022-07-31:
It happens the same with me! I was loosing my mind on it!
https://github.com/iRobotEducation/create3_docs/discussions/124

roni-kreinin, 2023-02-09:
The robot seems to be publishing a different transform from `base_link` to `bumper_front_center` than what is described in the urdf:

Robot `/tf_static`:
```
- header:
    stamp:
      sec: 1675981068
      nanosec: 277544281
    frame_id: base_link
  child_frame_id: bump_front_center
  transform:
    translation:
      x: 0.17499999701976776
      y: 0.0
      z: 0.039000000804662704
    rotation:
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
```

`robot_state_publisher` `/tf_static` from urdf:
```
transforms:
- header:
    stamp:
      sec: 1675981179
      nanosec: 75899446
    frame_id: base_link
  child_frame_id: bump_front_center
  transform:
    translation:
      x: 0.0
      y: 0.0
      z: 0.039200000000000006
    rotation:
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
```