iRobotEducation/create3_sim #180: requires pytest>=5.3

cipy, 2022-05-12:
**Describe the bug**
A clear and concise description of what the bug is.

at least on arm64 the installation will fail

unless you `sudo pip install pytest==5.3`

**To Reproduce**
Steps to reproduce the behavior:
1. Ubuntu 20.04 ARM64

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

alsora, 2022-05-13:
Hi, could you please provide the build steps that you used and the complete error that you got?

cipy, 2022-05-13:
I ran step by step from your Readme, I think this step failed with the `pytest` error

```
export IGNITION_VERSION=edifice
colcon build --symlink-install
source install/local_setup.bash
```

also I had to install `ros-rolling-desktop`, as galactic is not built for Ubuntu 22.04

cipy, 2022-05-13:
I think it might have been a silent error in `colcon`, but I remembered noticing at `rosdep`(?)  a pyTest version mismatch, which after I fixed with sudo pip, the `colcon` run become happy too 💯

alsora, 2022-05-13:
```
this step failed with an pytest error
```

I would need to see this error to understand what's the best approach to fix the problem.

martinerk0, 2022-10-14:
when doing `pip install colcon-common-extensions` on 20.04 it throws

`ERROR: pytest-rerunfailures 10.2 has requirement pytest>=5.3, but you'll have pytest 4.6.9 which is incompatible.`

while  

`sudo apt install python3-colcon-common-extensions`
 
installs okay.