iRobotEducation/create3_sim #182: The Create 3 Setup page doesn't work

MendelTheRobot, 2022-05-15:
The Setup page for the Create 3 robot, edu.irobot.com/create3-setup, doesn't work. You can't open the download from step 1, and the step 4 link doesn't work, rendering the Create 3 inoperable.

shamlian, 2022-05-15:
Hi there,
Please e-mail us at create@irobot.com to help resolve this problem; your Create 3 should not be "rendered inoperable" by performing any of these steps. I'm not sure what you mean by "can't open the download;" the file with an "swu" extension is for loading onto the robot. Download that file onto your computer for use later, then once you've connected to the robot's SSID/hotspot (triggered in step 2 and connected in step 3), the step 4 link will work.
Thanks.
--Steve

shamlian, 2022-05-15:
FWIW I've closed this particular issue here because this is for reporting bugs in the Create 3 simulator. For future issues, please use the forums linked at the bottom of the [original page you cited](edu.irobot.com/create3-setup).

closed 2022-05-15 by shamlian

shamlian, 2022-05-16:
I don't know if it will help or not, but there's a bunch of video tutorials [here](https://edu.irobot.com/learning-library/create-3-getting-started); this one [here](https://players.brightcove.net/6055873617001/932tH4ih8_default/index.html?videoId=6302449517001) is about initial setup.