iRobotEducation/create3_sim #188: TF Tree on different topic name

xkaraman, 2022-10-19:
**Describe the bug**
I don't know where to report this for now, so i will just report it to its digital twin.

I have the Create 3 robot and on i have configured it through its' webserver to have a namespace say `/robot1`. 

I can see the topics have indeed pushed down to it but one concern is that `/tf` and `/tf_static/` are also pushed down i.e `/robot1/tf` and `/robot1/tf_static`.

Generally as far as i know `/tf` and `/tf_static` should be on global namespace and the frame-names instead should be pushed to a different ie `/robot1/odom` frame.

I would like to know if my thinking is wrong and how i can keep the namespace but have my `tf` tree correctly, so i can publish other transforms like lidar or camera frame in relation to say `base_link` frame.

Available topics: `/robot1/tf` should be still `/tf` but with frames like `/robot1/odom` etc. ( Correct me if i am wrong)
```
/parameter_events
/robot1/battery_state
/robot1/cmd_audio
/robot1/cmd_lightring
/robot1/cmd_vel
/robot1/dock
/robot1/hazard_detection
/robot1/imu
/robot1/interface_buttons
/robot1/ir_intensity
/robot1/ir_opcode
/robot1/kidnap_status
/robot1/mobility_monitor/transition_event
/robot1/mouse
/robot1/odom
/robot1/robot_state/transition_event
/robot1/slip_status
/robot1/static_transform/transition_event
/robot1/stop_status
/robot1/tf
/robot1/tf_static
/robot1/wheel_status
/robot1/wheel_ticks
/robot1/wheel_vels
/rosout
```

xkaraman, 2022-10-20:
Ok, thanks for the link. Got the reasoning but can't see how it was selected without a concrete solution even from the ROS team.
So what's the best way to handle it as far as you know? Should i just drop the namespace for now?

justinIRBT, 2022-10-20:
If you had a multi-robot use case, you could subscribe to the separate topics `/robot1/tf` `/robot2/tf` then make a TransformBroadcaster with the frames renamed to `robot1/frame1` and `robot2/frame1`, etc... in the standard `tf` tree

xkaraman, 2022-10-21:
Ok thanks for the information and help!