iRobotEducation/create3_sim #191: Package 'irobot_create_gazebo_bringup' not found

carlitoselmago, 2022-11-16:
When running the launch command after following all steps
```
ros2 launch irobot_create_gazebo_bringup create3_gazebo_aws_small.launch.py
```

I get 
""
Package 'irobot_create_gazebo_bringup' not found: "package 'irobot_create_gazebo_bringup' not found, searching: ['/home/ubuntu/create3_ws/install/irobot_create_toolbox', '/home/ubuntu/create3_ws/install/irobot_create_description', '/home/ubuntu/create3_ws/install/irobot_create_control', '/opt/ros/galactic']"

""
Running on Ubuntu 20.04.5 LTS on a virtual machine

alsora, 2022-11-18:
Have you built the code from sources or installed from debian packages?
can you provide your steps to reproduce?

carlitoselmago, 2022-11-18:
installed from debian packages , today it actually worked, i believe I might have done something wrong while coping multiple lines of install commands at once.

closed 2022-11-18 by carlitoselmago