iRobotEducation/create3_sim #194: Can't activate controller 'diffdrive_controller' with Ignition

fmrico, 2022-12-17:
**Describe the bug**

The simulation in Ignition fails to activate `diffdrive_controller`, so the robot does not move when sending commands throught `/cmd_vel`. The relevant logs messages, IMHO:

```
[ruby $(which ign) gazebo-1] [ERROR] [1671266582.459760634] [joint_state_broadcaster]: None of requested interfaces exist. Controller will not run.
[ruby $(which ign) gazebo-1] [WARN] [1671266582.459990877] []: Error occurred while doing error handling.
[ruby $(which ign) gazebo-1] [ERROR] [1671266582.460153565] [controller_manager]: After activating, controller 'joint_state_broadcaster' is in state 'unconfigured', expected Active
```
```
[ruby $(which ign) gazebo-1] [INFO] [1671266583.009923320] [controller_manager]: Configuring controller 'diffdrive_controller'
[ruby $(which ign) gazebo-1] [ERROR] [1671266583.029269717] [controller_manager]: Can't activate controller 'diffdrive_controller': Command interface with 'left_wheel_joint/velocity' does not exist
[spawner-29] [INFO] [1671266583.032560457] [spawner_diffdrive_controller]: Configured and activated diffdrive_controller
[ruby $(which ign) gazebo-1] [WARN] [1671266583.043478061] [diffdrive_controller]: Can't accept new commands. subscriber is inactive
[ruby $(which ign) gazebo-1] [WARN] [1671266583.148753400] [diffdrive_controller]: Can't accept new commands. subscriber is inactive
[ruby $(which ign) gazebo-1] [WARN] [1671266583.228742312] [diffdrive_controller]: Can't accept new commands. subscriber is inactive
```

**To Reproduce**
Steps to reproduce the behavior:
1. Launch `ros2 launch irobot_create_ignition_bringup create3_ignition.launch.py`
2. Press the Play button

**Expected behavior**
Moving robot

alsora, 2022-12-17:
Hi @fmrico, is this with ROS 2 Galactic or ROS 2 Humble?
I'm asking because I saw your other PR for the `asoragna/ros2-humble` branch and because this is very similar to the error I got the last time I tried that branch (and the reason why Humble is still on that dev branch).

alsora, 2022-12-19:
@fmrico can you please confirm which ROS 2 version is causing this problem? Is it Galactic or Humble?
I just tried the Create 3 simulator with ROS 2 humble and at first I had the same issue, but then I upgraded all the ROS 2 libraries
```
sudo apt-get update && sudo apt-get upgrade
```

and the issue was gone.

fmrico, 2022-12-21:
Hi @alsora 

I am using Humble, and the `asoragna/humble` branch. I have just upgraded and built again the workspace, and I am still having the same problem.

alsora, 2022-12-21:
Sorry @fmrico, I just realized that your error is with Gazebo ignition and not classic.

I confirm that there are issues with ignition.
In my log I see
```
[ruby $(which ign) gazebo-1] [Err] [SystemLoader.cc:78] Failed to load system plugin [ign_ros2_control-system.so] : couldn't find shared library.
```
and then 

```
[spawner-29] [INFO] [1671656263.191168466] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-29] [INFO] [1671656265.199836139] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-29] [INFO] [1671656267.209131499] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-29] [ERROR] [1671656269.217180802] [spawner_diffdrive_controller]: Controller manager not available
[ERROR] [spawner-29]: process has died [pid 4274, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner diffdrive_controller -c /controller_manager --ros-args --params-file /root/create3_ws/install/irobot_create_control/share/irobot_create_control/config/control.yaml'].
```

This seems the same issue described in https://github.com/ros-controls/gz_ros2_control/issues/89 and https://github.com/ros-controls/gz_ros2_control/issues/91

alsora, 2022-12-22:
Ok, I managed to get this working by building https://github.com/ros-controls/gz_ros2_control from sources using the `master` branch.

The complete instructions are as follow:
```
mkdir -p ~/ws/src
cd ~/ws/src
git clone -b asoragna/humble https://github.com/iRobotEducation/create3_sim
git clone -b master https://github.com/ros-controls/gz_ros2_control
cd ~/ws
rosdep install --from-path src -yi --rosdistro humble
IGNITION_VERSION=fortress colcon build
```

and then run with
```
ros2 launch irobot_create_ignition_bringup create3_ignition.launch.py
```

slowrunner, 2023-12-09:
YEAR LATER - NO RESOLUTION HERE OR ROS-CONTROL - Still blocking using ignition for create3_sim (as documented in the instructions.)  

Perhaps the work around above works, but using the current Create3_Sim documentation, like any new Create3 owner that I am, doesn't work - change the create3_sim README to eliminate discussing the Ignition option.

aharshac, 2024-07-23:
https://github.com/iRobotEducation/create3_sim/issues/220#issuecomment-2245717396
> Hello @slowrunner. Did you find a workaround?
> 
> I had the same (well I'm not a 100% sure) problem on Humble (Ubuntu 22.04) with Fast DDS. The same isn't found with Cyclone DDS.
> 
> Looks like there's an issue on ros2_control [ros-controls/ros2_control#1200](https://github.com/ros-controls/ros2_control/issues/1200) with a probable fix [ros-controls/ros2_control#1562](https://github.com/ros-controls/ros2_control/pull/1562) I was unable to build the fix branch locally, but I hope this PR will solve the problem.

aharshac, 2024-08-26:
> [#220 (comment)](https://github.com/iRobotEducation/create3_sim/issues/220#issuecomment-2245717396)
> 
> > Hello @slowrunner. Did you find a workaround?
> > I had the same (well I'm not a 100% sure) problem on Humble (Ubuntu 22.04) with Fast DDS. The same isn't found with Cyclone DDS.
> > Looks like there's an issue on ros2_control [ros-controls/ros2_control#1200](https://github.com/ros-controls/ros2_control/issues/1200) with a probable fix [ros-controls/ros2_control#1562](https://github.com/ros-controls/ros2_control/pull/1562) I was unable to build the fix branch locally, but I hope this PR will solve the problem.

Ignition works for me after the issue above was fixed!