iRobotEducation/create3_sim #195: Release for Humble and Rolling?

SteveMacenski, 2022-12-19:
As part of https://github.com/ros-planning/navigation2/issues/2997, we want to migrate away from the TB3 to the TB4 (or iCreate) and Gazebo to Ignition. To do so, we need them to be available in the distribution so users can pull it in with a Nav2 install and CI can turn over without adding more packages to our underlay list.

Are there plans to release binaries in the near-term future?

alsora, 2022-12-19:
Hi Steve, 
please note that my answer is only related to the Create3 side of the house.

For what concerns ROS 2 Humble:
 - We plan to release the create 3 simulator (this package) very soon (hopefully this week, unless some strange issues are discovered during testing).
 - We already have a beta release for the robot firmware; unfortunately I can't give you a date for the stable release there as the software is currently not passing yet our quality gates due to some crashes/deadlocks in the ROS 2 core libraries and an increase in CPU usage when enabling BLE mode.

For what concerns ROS 2 Rolling:
 - It's not clear if all the simulator dependencies (in particular those related to ROS 2 control) have a rolling release, but we could have a branch that can be built from sources and it's compatible with Rolling.
 - We don't currently plan to release robot firmwares compatible with ROS 2 Rolling. It's possible that a ROS 2 Humble robot will be able to communicate with a ROS 2 Rolling laptop.

SteveMacenski, 2022-12-19:
I don't really care too much about the firmware to be honest. This is fully for simulation for CI testing and default bringups for folks to have nice demos to start off with :smile: . I believe this is the simulation repo as well!

> It's not clear if all the simulator dependencies (in particular those related to ROS 2 control) have a rolling release, but we could have a branch that can be built from sources and it's compatible with Rolling.

This is a minimum requirement for us in Nav2-land to use it. We need binaries. If I can help unblock or have things released that you rely on, let me know. I suspect they are released in Rolling though, moveit2 is in rolling and they rely on ROS 2 control.

alsora, 2022-12-19:
Okok perfect.

Let me first focus on the ROS 2 Humble release and then I will get back to you with more details about ROS 2 Rolling and potentially missing packages.

SteveMacenski, 2022-12-19:
Ok! Thanks!

alsora, 2022-12-22:
@SteveMacenski, quick update on the ROS 2 Humble release.
We have a fully functional gazebo-classic simulation, but there are several problems in the ignition simulation due to the ROS 2 control libraries.
 - the first issue is that there isn't a debian package for ign_ros2_control in Humble (there is in rolling FWIW) https://github.com/ros-controls/gz_ros2_control/issues/82
 - even with a build from source, it looks like that there are problems with using that package as mentioned in https://github.com/ros-controls/gz_ros2_control/issues/89, https://github.com/ros-controls/gz_ros2_control/issues/91 and https://answers.ros.org/question/409267/how-to-activate-controller-in-ros2-control-with-gazebo/

SteveMacenski, 2022-12-22:
Ugh OK. I was hoping to do these all in the same thrust. Lets see if our comments will spark some movement. I'm surprised given Ignition's "the simulator of ROS 2" with Gazebo deprecated that this could still be an issue

alsora, 2022-12-23:
I did a quick test for a ROS 2 rolling release.
The rolling debian package for `ign_ros2_control` seems broken and it causes a segmentation fault in Gazebo.
I opened the following ticket https://github.com/ros-controls/gz_ros2_control/issues/105

Building from sources seems to fix the problem (at least the simulation does not immediately crash), but I didn't test it thoroughly.

We may release the Create 3 simulator regardless (Gazebo classic works just fine), and add a disclaimer that users will have to build `gz_ros2_control` from sources.
Eventually, when those issues are fixed, no changes will be needed on our side.

splouisliu, 2023-01-03:
@alsora 
Is it currently possible to use the simulator on Gazebo classic by building from the humble branch or do we have to wait for a release? So far, building is running into errors for me.

alsora, 2023-01-19:
Hi @splouisliu, I just created an official `humble` branch.
I tested it locally and gazebo classics builds and runs without problems for me

ruffsl, 2023-03-09:
Sounds like https://github.com/turtlebot/turtlebot4_simulator/issues/32#issuecomment-1458331570 is getting close to cutting a release for humble. Would that of course now be the case for `create3_sim` as well?

alsora, 2023-03-09:
We already did a first humble release of the package.
We should do another one now that we fixed some additional bugs/errors

alsora, 2023-03-10:
At the moment, Rolling builds fine but I haven't verified if the runtime issues have been resolved.
I can still release it if needed.

alsora, 2023-03-10:
I realized that before releasing this repo for rolling, we need also to release its dependency.
I opened PR here https://github.com/ros/rosdistro/pull/36434

SteveMacenski, 2024-05-29:
@alsora reviving this - can we get Jazzy/Rolling support? This is blocking the Nav2 release or we'll need to find another robot platform to base from

alsora, 2024-06-01:
I'll look into doing a Jazzy release.

martincerven, 2024-06-01:
> I'll look into doing a Jazzy release.

I guess for Jazzy You'll need to use Gazebo Harmonic, which [ditched ignition name](https://gazebosim.org/docs/harmonic/migration_from_ignition), that means rewrite of few sources  😃

alsora, 2024-06-15:
FYI:
 - i created a branch `jazzy` for the `create3_sim` and for the `irobot_create_msgs` repo
 - i started the process to release `irobot_create_msgs` for jazzy (in the meantime you need to build from sources)
 - the `jazzy` branch of the `create3_sim` builds fine and I don't see any error at startup, but it doesn't work yet
 
 Known issues:
  - the GUI window is all black
  - the robot doesn't move (e.g. requesting actions or teleoperating it)

It would be very appreciated if you could take a look and provide suggestions.

martincerven, 2024-06-15:
I forgot to mention that there was already some work in [Turtlebot4 repo](https://github.com/turtlebot/turtlebot4/issues/412#issuecomment-2134225900).

One suggestion from me would be to tagret [**Gazebo Harmonic**](https://gazebosim.org/docs/latest/ros_installation)  instead of Jazzy (maybe call the branch Jazzy, but it could run on Humble too). You can actually run Harmonic with Humble, but you have to build [ros_gz](https://github.com/gazebosim/ros_gz) from source.

One use case is **Nvidia Jetson**  devices which are using now older Ubuntu (22.04) with Humble. My _humble_ guess would be that a large share if not majority of roboticists using create3 would use Jetsons (Me included), so it would be beneficial if the Jazzy branch would run seamlessly with Humble using Gazebo Harmonic. 

I know this is sim repo, but the Jetsons are able to run Gazebo simulations pretty fast even though they are embedded SoCs (they have good graphic drivers). I could test the sim on Humble & Harmonic & AGX Orin, my only 24.04 Jazzy device is RPi 5 which is totally slow for any simulation 😁

alsora, 2024-06-16:
Thanks, I'll take a look as it seems that the user there already ported this repo to jazzy and harmonic.

About the naming and supported versions: the gazebo harmonic page recommends to use jazzy so I'm not going to advertise using a different version of ROS.

Note that the code builds without any ROS-specific changes, so I would expect that once the gazebo issues are fixed, you should be able to compile this "jazzy" branch from sources against humble ROS libraries

martincerven, 2024-06-16:
> Note that the code builds without any ROS-specific changes, so I would expect that once the gazebo issues are fixed, you should be able to compile this "jazzy" branch from sources against humble ROS libraries

Yeah I meant it like this. Of course there are recommended versions of Gazebo for each ROS version, but since they provide compatibility matrix at https://github.com/gazebosim/ros_gz/tree/humble for different ROS versions, they kinda expect people will going to mix the versions. (They also provide binaries for Humble+Harmonic, but at that point it's easier to build it from source I guess)

For me it would be jumping from Gazebo Classic to Harmonic without using Ignition at all.

Also 24.04 is not yet that stable, so more people would want to use Harmonic with 22.04 for example.

martincerven, 2024-06-21:
Any updates @alsora ? I'd like to help & contribute, but I don't want to redo any work you already did in local/private git repo.

alsora, 2024-06-21:
@martincerven no updates yet.
I don't have any private/local repo; all the work I did so far is pushed to the branch `jazzy` of this repo.

Any help is really appreciated.

martincerven, 2024-07-26:
I have started looking into it, there might be more bugs with [Harmonic](https://github.com/gazebosim/gz-sim/issues/2488) itself, but at least there is some progress

alsora, 2024-08-16:
FYI, thanks to @civerachb-cpr, the `jazzy` branch is now ready for use.
We'll keep testing it for a while and then I'll proceed with a release.

martincerven, 2024-08-25:
I built it inside 24.04 VM, and it runs normally, but since it's VM it doesn't have high OpenGL version so I have to change:
```
<!-- <engine>ogre2</engine> -->
<engine>ogre</engine> 
```
in `gui.config` and then I get these black & white colors 😃  (but everything runs fine)

[image]

Also, I haven't been able to run Humble+Harmonic with this jazzy branch, I build all packages from source but, the robot doesn't move as with 24.04 VM, maybe diff_drive controller expects different topic, or there are too many breaking changes in other packages 😑

 @civerachb-cpr   Have you, by any chance, tried to use Humble + Harmonic with create3?

civerachb-cpr, 2024-08-26:
@martincerven : between Humble and Jazzy there were some pretty significant changes from the `ros_control` community, the biggest of which was changing to stamped messages for teleop control. Instead of using `geometry_msgs/Twist` messages, the Jazzy branch is set up to use `geometry_msgs/TwistStamped` for the control inputs. If the rest of your environment is Humble, you may have other nodes that are publishing/subscribing to the unstamped messages. That could explain why the robot isn't driving.

I have not tried using Humble & Harmonic together. Officially that combination is classified as "Possible, but use with caution" (see: https://gazebosim.org/docs/latest/ros_installation/#summary-of-compatible-ros-and-gazebo-combinations). If you are using ROS 2 Humble, I would strongly recommend using Gazebo's Fortress release.