iRobotEducation/create3_sim #202: Can't Build create3_sim

retupmoc1, 2023-02-20:
The colcon build of create3_sim fails for me with cmake errors. I think I went thru the steps in the create3_sim readme pretty faithfully, but deviated on the ignition version before trying the build...

- When using the script in the readme to install ignition-edifice, I got this error:  "E: Unable to locate package ignition-edifice". I changed to ignition-fortress, which succeeded. This seemed ok because I believe fortress is the recommended version for Humble anyway.
- After cloning the humble branch of create3_sim into a newly-created ~/create3_ws/src folder and changing the export IGNITION_VERSION value from edifice to fortress, then the colcon build failed. This is the full text of the script and the output:

`export IGNITION_VERSION=fortress
colcon build --symlink-install
source install/local_setup.bash
Starting >>> irobot_create_toolbox
Starting >>> irobot_create_control
Starting >>> irobot_create_description
Starting >>> irobot_create_ignition_plugins
Starting >>> aws_robomaker_small_house_world
Finished <<< irobot_create_control [0.47s]                                 
Finished <<< irobot_create_description [0.48s]
--- stderr: aws_robomaker_small_house_world                                
CMake Error at /usr/local/lib/cmake/Boost-1.72.0/BoostConfig.cmake:120 (find_package):
  Could not find a configuration file for package "boost_program_options"
  that exactly matches requested version "1.72.0".

  The following configuration files were considered but not accepted:

    /usr/lib/x86_64-linux-gnu/cmake/boost_program_options-1.74.0/boost_program_options-config.cmake, version: 1.74.0
    /lib/x86_64-linux-gnu/cmake/boost_program_options-1.74.0/boost_program_options-config.cmake, version: 1.74.0
Call Stack (most recent call first):
  /usr/local/lib/cmake/Boost-1.72.0/BoostConfig.cmake:185 (boost_find_component)
  /usr/share/cmake-3.22/Modules/FindBoost.cmake:594 (find_package)
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:197 (find_package)
  /opt/ros/humble/share/gazebo_dev/cmake/gazebo_dev-extras.cmake:3 (find_package)
  /opt/ros/humble/share/gazebo_dev/cmake/gazebo_devConfig.cmake:41 (include)
  /opt/ros/humble/share/gazebo_ros/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/humble/share/gazebo_ros/cmake/gazebo_rosConfig.cmake:41 (include)
  CMakeLists.txt:4 (find_package)

Failed   <<< aws_robomaker_small_house_world [0.94s, exited with code 1]
Aborted  <<< irobot_create_toolbox [0.99s]
Aborted  <<< irobot_create_ignition_plugins [1.12s]                            

Summary: 2 packages finished [1.37s]
  1 package failed: aws_robomaker_small_house_world
  2 packages aborted: irobot_create_ignition_plugins irobot_create_toolbox
  1 package had stderr output: aws_robomaker_small_house_world
  8 packages not processed
`

I have only a little experience with cmake and have zero experience with boost, so this has me baffled. When I run this command:
`dpkg -s libboost-dev | grep 'Version'` the boost version returned is 1.74.0.3ubuntu7. However, the error call stack refers to /usr/local/lib/cmake where all the files and folders imply 1.7.2.

Other setup notes:

- Running ROS2 humble installed on a Ubuntu 22.04 laptop
- The installed Gazebo version is 11.10.2
- I have been able to list the create3 topics, control the machine with a joystick controller, etc., no problem

Any help would be greatly appreciated!

Thanks,
Steve

alsora, 2023-03-06:
Hi @retupmoc1, unfortunately it won't be easy to help you as this error seems related to your laptop/setup rather than a problem in this repository.
I can confirm that the repository builds as expected on a clean Ubuntu machine.

Let me start with a quick question though.
Is there a specific reason why you need to build this repository from sources?
The Create 3 simulator is released as some debian package and you can install it with
```
sudo apt-get update
sudo apt-get install ros-humble-irobot-create-ignition-sim
```
(or use `sudo apt-get install ros-humble-irobot-create-gazebo-sim` if you are interested in gazebo classic).

If you need to build from sources, it looks like there's a problem with your boost setup.
You may want to try to do
```
sudo apt-get update
sudo apt-get upgrade
sudo apt-get install libboost-dev
sudo apt-get install libboost-program-options-dev
```

retupmoc1, 2023-03-06:
Hi Alberto,

Understood on the local config. I was hoping someone had run into the same or similar issue. I'll try the shell commands you suggested not because I need to build from source, but because I'm now determined to make it work and might learn something in the process.

Also, I did not realize there was a package! Hah! Thanks for pointing this out. I'll keep an eye on the available package list with "irobot-create" in the name going forward.

Thanks,
Steve

slowrunner, 2023-12-09:
> Let me start with a quick question though. Is there a specific reason why you need to build this repository from sources? The Create 3 simulator is released as some debian package and you can [...]

Why does the README not suggest these instead of only a "build from Create3_sim sources"?  

Sorry not OP, but have spent over a day trying multiple times following the Create3_sim docs trying the Ignition approach - finally managed understand Humble Gazebo Classic Create3_sim works, Ignition not.