iRobotEducation/create3_sim #203: Incorrect timestamps published using Gazebo simulation

oldtopos, 2023-02-23:
**Describe the bug**
Running Gazebo and RViz, RViz reports an incorrect TF tree for links wheel_drop_left, left_wheel, wheel_drop_right, and right_wheel

**To Reproduce**
Steps to reproduce the behavior:
1. Build or install ROS2 Rolling and Classic Gazebo
2. Build create3_sim and associated packages
3. ros2 launch irobot_create_gazebo_bringup create3_gazebo.launch.py
4. Examine RViz/RobotModel/Links for errors
5. Verify incorrect timestamps using ros2 topic info -v /joint_states
6. Observe ros2 param get joint_state_publisher use_sim_time returns False

**Expected behavior**
ros2 param get joint_state_publisher use_sim_time returns True
ros2 topic info -v /joint_states reports consistent sim based timestamps rather than some using system time

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add
        parameters=[
            {'use_sim_time': True},
        ]
after line 48 of robot_description.launch.py, rebuild, and re-execute launch file to correct errors

alsora, 2023-03-08:
Thank you for reporting.
The issue has been fixed as part of this PR https://github.com/iRobotEducation/create3_sim/pull/200

closed 2023-03-08 by alsora

schenkch, 2023-06-01:
I have the same issue but with running turtlebot4_navigation (with the physical robot) and rviz in Galactic. I tried updating the files as it was done here (https://github.com/turtlebot/turtlebot4/commit/7f6d8f0b5e348024d324346f2e0c4a9d31af8240) for Humble. How can I make sure that these changes actually take effect? How do I rebuild the code? It would be great if somebody can help me out.