iRobotEducation/create3_sim #208: SEGFAULT when adding a second Create3 Robot

Gior26, 2023-03-31:
**Describe the bug**
Using the roni-kreinin/multi_robot fork from the pull request #207 the gazebo server encounters a SEGFAULT error whenever I try to add a second Create3 unit.

**To Reproduce**
Steps to reproduce the behavior:
1. ros2 launch irobot_create_ignition_bringup ignition.launch.py
2. ros2 launch irobot_create_ignition_bringup create3_spawn.launch.py namespace:=robot1
3. ros2 launch irobot_create_ignition_bringup create3_spawn.launch.py namespace:=robot2 x:=3.0
4. SEGFAULT on Gazebo server

**Expected behavior**
A second Create3 unit should spawn inside the simulation.

**Additional context**
- OS: Ubuntu 22.04.2
- Gazebo: Ignition Fortress
- ROS: Ros Humble

The first unit works correctly with the /robot1 namespace

[gazebo-log.txt](https://github.com/iRobotEducation/create3_sim/files/11123122/gazebo-log.txt)
[robot1-log.txt](https://github.com/iRobotEducation/create3_sim/files/11123125/robot1-log.txt)
[robot2-log.txt](https://github.com/iRobotEducation/create3_sim/files/11123126/robot2-log.txt)
[gdb-stack.txt](https://github.com/iRobotEducation/create3_sim/files/11123136/gdb-stack.txt)

roni-kreinin, 2023-03-31:
@Gior26 You probably need to update your `ign_ros2_control` package. Version 0.4.4 fixes this issue.

Gior26, 2023-04-01:
Thanks, I've missed that dependency now everything works correctly.
I'm closing the issue

closed 2023-04-01 by Gior26