iRobotEducation/create3_sim #209: Ignition: Missing IR intensity sensors

Gior26, 2023-04-12:
**Describe the bug**
Using Gazebo Ignition under the /ir_intensity topic data from the 7 sensors is missing.
Only data from the base_link frame is available and it is positioned on the left side of the robot. 

**To Reproduce**
Steps to reproduce the behavior:
1. ros2 launch irobot_create_ignition_bringup create3_ignition.launch.py
2. Add an object with collision
3. ros2 topic echo /ir_intensity
4. Only IR data from base_link is available

**Expected behavior**
Under /ir_intensity data from all 7 IR sensors should be available

**Screenshots**
[image]

**Additional context**
- Passing argument gazebo:=ignition didn't solve the issue.
- IR sensors works with gazebo classic.
- OS: Ubuntu 22.04.2
- Gazebo: Ignition Fortress
- ROS: Ros Humble

[Libraries](https://github.com/iRobotEducation/create3_sim/files/11209737/libraries.txt) and [log](https://github.com/iRobotEducation/create3_sim/files/11209762/log.txt)

alsora, 2023-04-12:
If you do `ros2 topic list --include-hidden` you should see the following topics
https://github.com/iRobotEducation/create3_sim/blob/main/irobot_create_ignition/irobot_create_ignition_bringup/config/sensors_params.yaml#L16-L23

These topics contain the "raw" IR data that is then fused in the `ir_intensity` topic.
Can you try to echo them ? This should help understand where the problem is.

Gior26, 2023-04-12:
Here's the log for [_internal/ir_intensity_front_center_left/scan](https://github.com/iRobotEducation/create3_sim/files/11210204/ir_intensity_front_left_center_scan.txt) and [_internal/ir_intensity_front_center_left](https://github.com/iRobotEducation/create3_sim/files/11210207/ir_intensity_front_left_center.txt).

I'm noticing that timestamp inside _internal/ir_intensity_front_center_left are always 0, I don't know if that may be the issue.

Gior26, 2023-04-12:
Here's the output from all the _internal/ir_intensity_xxx/scan
-  [ir_intensity_front_center_left_scan.txt](https://github.com/iRobotEducation/create3_sim/files/11212703/ir_intensity_front_center_left_scan.txt)
-  [ir_intensity_front_center_right_scan.txt](https://github.com/iRobotEducation/create3_sim/files/11212705/ir_intensity_front_center_right_scan.txt)
- [ir_intensity_front_left_scan.txt](https://github.com/iRobotEducation/create3_sim/files/11212707/ir_intensity_front_left_scan.txt)
- [ir_intensity_front_right_scan.txt](https://github.com/iRobotEducation/create3_sim/files/11212711/ir_intensity_front_right_scan.txt)
- [ir_intensity_left_scan.txt](https://github.com/iRobotEducation/create3_sim/files/11212712/ir_intensity_left_scan.txt)
- [ir_intensity_right_scan.txt](https://github.com/iRobotEducation/create3_sim/files/11212714/ir_intensity_right_scan.txt)
- [ir_intensity_side_left_scan.txt](https://github.com/iRobotEducation/create3_sim/files/11212715/ir_intensity_side_left_scan.txt)

paoloelle, 2025-02-28:
Hi @Gior26 did you solve this issue?

Gior26, 2025-02-28:
> Hi [@Gior26](https://github.com/Gior26) did you solve this issue? 

No, I've moved on using another robot with a 2D Lidar.

closed 2025-02-28 by Gior26