iRobotEducation/create3_sim #212: Sensor Issue with a namespaced multi_sim

MatteoCarlone, 2023-05-19:
ROS2 Humble,
Ubuntu 22.04,

I'm using the last released multi-robot simulation in the main branch.
I'm able to spawn more than one create correctly namespaced.
I had the necessity of adding a lidar sensor on top of the creates in order to use the NAV2 localization and navigation.
I just added a xacro lidar description to the create urdf and namespaced it in the plugin section, I report the script below:
```XML
<?xml version="1.0" encoding="utf-8"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">

    <xacro:macro name="lidar_sensor" params="gazebo namespace">

        <material name="black">
            <color rgba="0 0 0 1"/>
        </material>

        <link name="lidar_link">
            <inertial>
                <mass value="0.0001"/>
                <origin xyz="0 0 0"/>
                <inertia ixx="0.0001" ixy="0.0" ixz="0.0" iyy="0.0001" iyz="0.0" izz="0.0001"/>
            </inertial>
            <visual>
                <origin xyz="0 0 0.008" rpy="0 0 1.57"/>
                <geometry>
                    <mesh filename="package://irobot_create_description/meshes/turtlebot4/rplidar.dae"/>
                </geometry>
                <material name="black"/>
            </visual>
        </link>

        <joint name="lidar_joint" type="fixed">
            <parent link="base_link"/>
            <child link="lidar_link"/>
            <origin xyz="-0.05 0 0.175" rpy="0 0 0"/>
        </joint>
        
        <gazebo reference="lidar_joint">
    		<preserveFixedJoint>true</preserveFixedJoint>	
  	    </gazebo>

        <gazebo reference="lidar_link">
            <material>Gazebo/Black</material>
            <sensor type="ray" name="lidar_link">
                <always_on>true</always_on>
                <visualize>false</visualize>
                <pose>0.0 0 0 0 0 0</pose>
                <update_rate>10</update_rate>
                <ray>
                    <scan>
                        <horizontal>
                            <samples>360</samples>
                            <resolution>1.000000</resolution>
                            <min_angle>-3.14</min_angle>
                            <max_angle>3.14</max_angle>
                        </horizontal>
                    </scan>
                    <range>
                        <min>0.3</min>
                        <max>12.0</max>
                        <resolution>0.01</resolution>
                    </range>
                    <noise>
                        <type>gaussian</type>
                        <mean>0.0</mean>
                        <stddev>0.01</stddev>
                    </noise>
                </ray>
                <xacro:if value="${gazebo == 'classic'}">
    		        <plugin name="gazebo_lidar" filename="libgazebo_ros_ray_sensor.so">
    		            <ros>
    		                <namespace>${namespace}</namespace>
    		                <remapping>~/out:=scan</remapping>
    		            </ros>
                        <output_type>sensor_msgs/LaserScan</output_type>
                        <frame_name>lidar_link</frame_name>
    		        </plugin>
                </xacro:if>
            </sensor>
        </gazebo>
    </xacro:macro>
</robot>

```

and in the create3 description just:
```XML
<xacro:include filename="$(find irobot_create_description)/urdf/sensors/lidar_sensor.xacro"/>
```
```XML
<xacro:lidar_sensor gazebo="$(arg gazebo)" namespace="$(arg namespace)"></xacro:lidar_sensor>
```

**I have the following issue:**

The first create that I spawn has the lidar with the `/robot1/scan` topic visible and working correctly, once I add the second robot to the simulation:
```Bash
ros2 launch irobot_create_gazebo_bringup create3_spawn.launch.py namespace:=robot2 x:=1.0
```
it spawn with the lidar visuals but there's no `/robot2/scan` topic published. 

I Know that's probably not a create3 issue, I was just wondering if you solved this while creating the Turtlebot4 urdf description. 
Thanks in advance for your help, if it's off-topic I'll close this.

MatteoCarlone, 2023-05-26:
@roni-kreinin, may you please give me a hint about this? I'm quite sure you already deal with this with the turtlebot4.
This would really help me thanks again!

roni-kreinin, 2023-05-30:
@MatteoCarlone Are you using gazebo classic or ignition?

MatteoCarlone, 2023-05-30:
Gazebo Classic sorry that I haven’t specified it in the previous question!

roni-kreinin, 2023-05-30:
TurtleBot4 only supports Ignition so I never tried setting up the RPLIDAR in classic. Your approach does look correct though.

Does the same thing happen if you spawn the first robot without a namespace?

MatteoCarlone, 2023-05-30:
@roni-kreinin I just tried this:

*First robot without namespace*
```
ros2 launch irobot_create_gazebo_bringup create3_gazebo.launch.py 
```
*second robot with namespace robot1*
```
ros2 launch irobot_create_gazebo_bringup create3_spawn.launch.py namespace:=robot1 y:=3.0
```

I get the same problem, I just have the `/scan` topic not namespace that publish two times:
```
ros2 topic info /scan --verbose

Type: sensor_msgs/msg/LaserScan

Publisher count: 2

Node name: scan
Node namespace: /
Topic type: sensor_msgs/msg/LaserScan
Endpoint type: PUBLISHER
GID: 01.0f.5e.ca.0e.17.f3.8d.01.00.00.00.00.00.76.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Node name: scan
Node namespace: /
Topic type: sensor_msgs/msg/LaserScan
Endpoint type: PUBLISHER
GID: 01.0f.5e.ca.0e.17.f3.8d.01.00.00.00.00.02.43.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

```