iRobotEducation/create3_sim #217: Disable Reflexes

msminirobot, 2023-11-20:
**Describe the bug**
According to the Create 3 Docs, reflexes can be enabled or disabled using the corresponding ROS 2 parameters exposed by the motion_control ROS 2 node. However, the update of the parameter always fails when using ros2 parameter set /motion_control reflexes_enabled false. 

**To Reproduce**
Steps to reproduce the behavior:
1. Launch a simulation of the Create 3 (e.g., the TB4 simulation here: ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py)
2. Open a second terminal 
3. Execute ros2 parameter set /motion_control reflexes_enabled False 
4. See error

**Expected behavior**
Reflexes_enabled set to False 

**Additional context**
According to [this line in Code](https://github.com/iRobotEducation/create3_sim/blob/6a8bc44f9be2c260bc17e753c31455d6759af238/irobot_create_common/irobot_create_nodes/src/motion_control_node.cpp#L194), set parameters always returns false. In contrast to the comment, reflexes are enabled and executed and cannot be turned off.

alsora, 2023-11-20:
Good catch. This is a bug.
The simulator didn't initially implement reflexes, so we forced that parameter to return false.
Then we added reflexes, but we forgot to update the parameter logic.
It should be an easy fix, I'll try to put up a PR in the next few days.

msminirobot, 2023-11-20:
Thanks a lot :)

jonaskuResearch, 2025-05-16:
This seems to still be an issue, unfortunately. Are there any plans to implement this?