iRobotEducation/create3_sim #220: Remove the Ignition install option from README (Will Not Run Create3_Sim)

slowrunner, 2023-12-09:
For over a year it appears that new Create3 users following the Create3_sim README and choosing to install Ignition Gazebo have discovered **IT DON'T WORK!**.  

(The issue was reported here, and with [ROS-Control over a year ago](https://github.com/ros-controls/gz_ros2_control/issues/89), but there has been no action.)

Please remove the Ignition install option from the README until the issue is resolved.

**To Reproduce**
1. [Follow the Ignition install](https://github.com/iRobotEducation/create3_sim#ignition-fortress)
2. Launch: ros2 launch irobot_create_ignition_bringup create3_ignition.launch.py
3. Controller_Manager dies - [Issue 194](https://github.com/iRobotEducation/create3_sim/issues/194):
```
1702090266.5477965 [ERROR] [spawner-25]: process has died [pid 10052, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spaw
ner joint_state_broadcaster -c controller_manager --ros-args -r __ns:=/'].
1702090266.5570664 [INFO] [spawner-38]: process started with pid [10445]
1702090277.5307651 [ERROR] [spawner-38]: process has died [pid 10445, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spaw
ner diffdrive_controller -c controller_manager --ros-args -r __ns:=/ --params-file /home/ubuntu/wali_desk/c3ws/install/irobot_creat
e_control/share/irobot_create_control/config/control.yaml'].
```

(Or I guess if [the work-around](https://github.com/iRobotEducation/create3_sim/issues/194#issuecomment-1362670982) succeeds, add that to the README instead of removing Ignition option)

I'm so frustrated chasing this, reinstalling, trying different things I could think of and wasting a whole day on a "known for over a year" black hole.

**Expected behavior**  Working Create3 Simulation in Ignition Freedom

aharshac, 2024-07-23:
Hello @slowrunner. Did you find a workaround?

I had the same (well I'm not a 100% sure) problem on Humble (Ubuntu 22.04) with Fast DDS. The same isn't found with Cyclone DDS.

Looks like there's an issue on ros2_control https://github.com/ros-controls/ros2_control/issues/1200 with a probable fix https://github.com/ros-controls/ros2_control/pull/1562
I was unable to build the fix branch locally, but I hope this PR will solve the problem.

slowrunner, 2024-07-23:
> Hello @slowrunner. Did you find a workaround?

No - stuck with classic.

destogl, 2024-08-16:
> Hello @slowrunner. Did you find a workaround?
> 
> I had the same (well I'm not a 100% sure) problem on Humble (Ubuntu 22.04) with Fast DDS. The same isn't found with Cyclone DDS.
> 
> Looks like there's an issue on ros2_control [ros-controls/ros2_control#1200](https://github.com/ros-controls/ros2_control/issues/1200) with a probable fix [ros-controls/ros2_control#1562](https://github.com/ros-controls/ros2_control/pull/1562) I was unable to build the fix branch locally, but I hope this PR will solve the problem.

We have fixed the issues in all supported distros. Still not released, but can be tested at least.

aharshac, 2024-08-26:
@destogl It works now. Thanks for the fix!

slowrunner, 2024-08-26:
Closing based on other's testing.

closed 2024-08-26 by slowrunner