iRobotEducation/create3_sim #225: Ignition: missing /cliff_intensity topic

paoloelle, 2024-02-08:
**Describe the bug**
Using Gazebo Ignition the topic named /cliff_intensity is missing.

**To Reproduce**
Steps to reproduce the behavior:
1. `ros2 launch irobot_create_ignition_bringup create3_ignition.launch.py`
2. `ros2 topic echo`
3. No topic named /cliff_intensity

**Expected behavior**
I should see a topic named  /cliff_intensity as in the real robot which provides reading from the ir cliff sensors.

**Additional context**

- Gazebo Ignition
- Ubuntu 22.04
- ROS Humble

Prince25, 2025-04-01:
I may be wrong, but I believe the topic name is different. Look for `/ir_intensity` instead.

paoloelle, 2025-04-02:
Hi @Prince25, Thanks for the reply. I had an error on my side (most probably), I forgot to close the issue. Thanks for pointing this out.

closed 2025-04-02 by paoloelle