iRobotEducation/create3_sim #226: Issue with Spawner for diffdrive_controller: Controller manager not available

zahra370, 2024-03-01:
encountered the "Controller manager not available" error when using the spawner for diffdrive_controller, 
error :   
[ERROR] [spawner-9]: process has died [pid 26229, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner joint_state_broadcaster -c controller_manager --ros-args -r __ns:=/'].
[INFO] [spawner-19]: process started with pid [26646]
[spawner-19] [INFO] [1709295965.721752111] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-19] [INFO] [1709295967.740792119] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-19] [INFO] [1709295969.760746372] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-19] [INFO] [1709295971.781857527] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-19] [ERROR] [1709295973.798772303] [spawner_diffdrive_controller]: Controller manager not available
[ERROR] [spawner-19]: process has died [pid 26646, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner diffdrive_controller -c controller_manager --ros-args -r __ns:=/ --params-file /home/zahra/turtbot_ws/install/irobot_create_control/share/irobot_create_control/config/control.yaml'].

BabeJoan, 2024-04-05:
Same problem, have you solved it?

nisarganc, 2024-04-19:
I've been encountering the same problem with spawner for joint_state_broadcaster. The issue seems to be from gazebo ignition plugin for differential drive controller.

alsora, 2024-09-19:
@civerachb-cpr do you think that your latest PR should fix this issue?
https://github.com/iRobotEducation/create3_sim/pull/233

civerachb-cpr, 2024-09-19:
It might, though this issue predates Jazzy. The issue that #233 fixes appears to be specific to a new update to ros_controllers; when the Jazzy simulation was first released the timeout wasn't an issue, so something's changed between then and now on the controller front.

I suspect #233 won't hurt, but it may not completely solve the problem either.