iRobotEducation/create3_sim #227: Building project gets stuck on 'colcon build --symlink-install' command

redealexander9, 2024-03-19:
**Describe the bug**
When building the project, running the command 'colcon build --symlink-install' freezes after it has finished processing the first 3 modules, and it slows down when it tries to process 4 at a time. After that it outputs "c++: fatal error: Killed signal terminated program cc1plus compilation terminated.
gmake[2]: *** [Create3Hmi/CMakeFiles/Create3Hmi.dir/build.make:85: Create3Hmi/CMakeFiles/Create3Hmi.dir/Create3Hmi_autogen/mocs_compilation.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:153 Create3Hmi/CMakeFiles/Create3Hmi.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< irobot_create_ignition_plugins [3min 18s, exited with code 2]"

**To Reproduce**
Steps to reproduce the behavior:
1. export IGNITION_VERSION=fortress
2. colcon build --symlink-install
3. See error

**Expected behavior**
The project should finish building and succeed

**Additional context**
Gazebo 11
Ubuntu 22.04
ROS humble

alsora, 2024-03-19:
Hi @redealexander9 
The compiler is crashing, likely because the platform where you are compiling is running out of resources.

Do you really need to build this package?
It can be installed via 
```
sudo apt-get update && sudo apt-get install -y \
        ros-humble-irobot-create-msgs \
        ros-humble-irobot-create-ignition-sim \
        ros-humble-irobot-create-gazebo-sim \
        ros-humble-irobot-create-nodes
```

If you really need to build it, you should limit the number of concurrent jobs
```
colcon build --executor sequential --symlink-install
```

redealexander9, 2024-03-19:
Thank you, I didn't know I could install it and not have to build it. That worked

closed 2024-03-19 by redealexander9