iRobotEducation/create3_sim #234: Issue with rosdep definition of gazebo

tillb98, 2024-10-13:
**Describe the bug**
Hello, I tried to install the Create Simulator by following the installation instructions step by step with a fresh Ubuntu 22.04 LTS Jammy and ROS Humble. When building the create3_ws, I encounter the following error after running “rosdep install --from-path src -yi”:

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
irobot_create_gz_bringup: Cannot locate rosdep definition for [gz_ros2_control]
irobot_create_toolbox: Cannot locate rosdep definition for [gz_math_vendor]
irobot_create_gz_plugins: Cannot locate rosdep definition for [gz_gui_vendor]

resolved into system dependencies:
irobot_create_gz_bringup: rosdep definition for [gz_ros2_control] not found
irobot_create_toolbox: rosdep definition for [gz_math_vendor] not found
irobot_create_gz_plugins: rosdep definition for [gz_gui_vendor] not found

As a result, the next step “colcon build --symlink-install” also fails. This results in the following output:
Starting >>> irobot_create_gz_plugins
--- stderr: irobot_create_gz_plugins 
CMake Error at CMakeLists.txt:14 (find_package):
By not providing "Findgz_gui_vendor.cmake" in CMAKE_MODULE_PATH this
project has asked CMake to find a package configuration file provided by
"gz_gui_vendor", but CMake did not find one.

Could not find a package configuration file provided by "gz_gui_vendor"
with any of the following names:

gz_gui_vendorConfig.cmake
gz_gui_vendor-config.cmake

Add the installation prefix of "gz_gui_vendor" to CMAKE_PREFIX_PATH or set
"gz_gui_vendor_DIR" to a directory containing one of the above files. If
"gz_gui_vendor" provides a separate development package or SDK, be sure it
has been installed.
---
Failed <<< irobot_create_gz_plugins [1.04s, exited with code 1]

What is the reason for this and how can I fix the problems with gz_ros2_control or gz_math_vendor and gt_gui_vendor? Please excuse the question, I am still relatively inexperienced with ROS. Many greetings and thank you!

**To Reproduce**
Follow the installation instructions step by step.

**Expected behavior**
Successful addition of dependencies and subsequent successful building of the workspace.

**Additional context**
Add any other context about the problem here.
[image]

tillb98, 2024-10-19:
> To install the create 3 simulator you should use:
> 
> ```
> sudo apt-get install ros-humble-irobot-create-gazebo
> ```
> 
> or
> 
> ```
> sudo apt-get install ros-humble-irobot-create-ignition
> ```
> 
> depending on which version of the simulator you want to use.

Hey alsora, I noticed that the command you suggested, “sudo apt search ros-humble-irobot-create-gazebo”, still requires the ls extensions “-sim”, “-bringup” or “-dbgsym”. Installing these packages with “sudo apt-get install” will then work. The package readme file may need to be updated because the prompts now seem to vary slightly (e.g. gazebo is now spelled out instead of gz), but the simulation seems to work. Thanks for your help!

rickarmstrong, 2024-12-07:
I noticed that if I do this (i.e. add a glob):
`sudo apt-get install ros-humble-irobot-create-gazebo*`

`apt` attempts to install four packages, but fails because of a conflict with `gz_tools`:
```
$sudo apt-get install ros-humble-irobot-create-gazebo*
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'ros-humble-irobot-create-gazebo-sim' for glob 'ros-humble-irobot-create-gazebo*'
Note, selecting 'ros-humble-irobot-create-gazebo-plugins-dbgsym' for glob 'ros-humble-irobot-create-gazebo*'
Note, selecting 'ros-humble-irobot-create-gazebo-plugins' for glob 'ros-humble-irobot-create-gazebo*'
Note, selecting 'ros-humble-irobot-create-gazebo-bringup' for glob 'ros-humble-irobot-create-gazebo*'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 gz-tools2 : Conflicts: gazebo (>= 11.0.0) but 11.10.2+dfsg-1 is to be installed
             Conflicts: gazebo (<= 11.14.0) but 11.10.2+dfsg-1 is to be installed
             Conflicts: gazebo:arm64 (<= 11.14.0)
             Conflicts: gazebo:arm64 (>= 11.0.0)
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.

```

rickarmstrong, 2024-12-07:
I think the real issue here is that the `jazzy` branch has a typo in the README: it calls-out ROS `humble` as a dependency. If I follow the instructions using Ubuntu 24.04 `noble` + ROS 2 `jazzy` + Gazebo `harmonic`, this repo builds as expected. 

P.S. After building, there's an issue launching the simulation, but that's a different problem having to do with `ros_control`. Workaround is to start the simulation one part at a time, `sim.launch.py`, start the simulation by hitting the run button in Gazebo, `create3_spawn.launch.py`, then `create3_gz_nodes.launch.py`.

tobias-kausch, 2025-01-13:
In case it is helpful to anyone, I resolved a similar error by making sure I was cloning the correct branch of this repository. In my case, I was cloning the repository using `git clone <repo_url>`. Using `git clone -b <branch_name> --single-branch <repo_url>` and proceeding with the rest of the guide worked without error.

>  When building the create3_ws, I encounter the following error after running “rosdep install --from-path src -yi”:
> 
> ERROR: the following packages/stacks could not have their rosdep keys resolved
> to system dependencies:
> irobot_create_gz_bringup: Cannot locate rosdep definition for [gz_ros2_control]
> irobot_create_toolbox: Cannot locate rosdep definition for [gz_math_vendor]
> irobot_create_gz_plugins: Cannot locate rosdep definition for [gz_gui_vendor]
> 
> As a result, the next step “colcon build --symlink-install” also fails. This results in the following output:
> Starting >>> irobot_create_gz_plugins
> --- stderr: irobot_create_gz_plugins
> CMake Error at CMakeLists.txt:14 (find_package):
> By not providing "Findgz_gui_vendor.cmake" in CMAKE_MODULE_PATH this
> project has asked CMake to find a package configuration file provided by
> "gz_gui_vendor", but CMake did not find one.

Dylan-B-Johnson, 2025-02-18:
> To install the create 3 simulator you should use:
> 
> ```
> 
> sudo apt-get install ros-humble-irobot-create-gazebo
> 
> ```
> 
> or 
> 
> ```
> 
> sudo apt-get install ros-humble-irobot-create-ignition
> 
> ```
> 
> 
> 
> depending on which version of the simulator you want to use.

A quick search [here](https://packages.ubuntu.com) shows that neither package exists anymore.

wjqgitdummy, 2025-02-23:
> Apparently, it's a slog to get ROS `humble` + Gazebo `harmonic` to play nice together. There's another open [issue](https://github.com/iRobotEducation/create3_sim/issues/195) discussing this in more detail.
> 
> P.S. After building, there's an issue launching the simulation, but that's a different problem having to do with `ros_control`. Workaround is to start the simulation one part at a time, `sim.launch.py`, start the simulation by hitting the run button in Gazebo, `create3_spawn.launch.py`, then `create3_gz_nodes.launch.py`.

Hi @rickarmstrong , did you manage to get create3_sim to work with humble + harmonic setup?

Dylan-B-Johnson, 2025-02-23:
> Hi @rickarmstrong , did you manage to get create3_sim to work with humble + harmonic setup?

I also tried humble + harmonic, but when I build I get the following:

CMake Error at CMakeLists.txt:14 (find_package):
  By not providing "Findangles.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "angles", but
  CMake did not find one.

  Could not find a package configuration file provided by "angles" with any
  of the following names:

    anglesConfig.cmake
    angles-config.cmake

  Add the installation prefix of "angles" to CMAKE_PREFIX_PATH or set
  "angles_DIR" to a directory containing one of the above files.  If "angles"
  provides a separate development package or SDK, be sure it has been
  installed.

Dylan-B-Johnson, 2025-02-23:
> Hi @rickarmstrong , did you manage to get create3_sim to work with humble + harmonic setup?

I also tried installing on humble + harmonic. After manually installing a bunch of packages to fix CMake errors, I got the following, which I could not get past:

CMake Error at /home/username/ros2_humble/ros2-linux/share/control_msgs/cmake/ament_cmake_export_libraries-extras.cmake:48 (message):
  Package 'control_msgs' exports the library
  'control_msgs__rosidl_generator_c' which couldn't be found
Call Stack (most recent call first):
  /home/username/ros2_humble/ros2-linux/share/control_msgs/cmake/control_msgsConfig.cmake:41 (include)
  CMakeLists.txt:15 (find_package)

rickarmstrong, 2025-02-23:
> Hi [@rickarmstrong](https://github.com/rickarmstrong) , did you manage to get create3_sim to work with humble + harmonic setup?

I ended up using https://github.com/ros-navigation/nav2_minimal_turtlebot_simulation in an `lxd` container running `noble` + `jazzy`.