iRobotEducation/create3_sim #235: Issue with rendering engine in the urdf

suchetanrs, 2024-10-27:
**Describe the bug**
The RPLidar simulation in turtlebot4 does not work well if the render engine is set to ogre. If it is changed to ogre2, it starts to work properly.

**To Reproduce**
Steps to reproduce the behavior:
Follow the setup instructions here https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_simulator.html

Run ros2 launch turtlebot4_gz_bringup turtlebot4_gz.launch.py

Add the visualize lidar plugin on gazebo and visualise the rplidar topic.

**Expected behavior**
Obstacles should be detected well.

**Screenshots**
With ogre render engine
[image]

With ogre2 render engine
[image]

suchetanrs, 2024-10-27:
Should I raise a PR for this with the render engine changed to ogre2? 
If yes, to which target branch should I raise one?

Thanks!

alsora, 2024-10-28:
@suchetanrs this appears to be an issue with the turtlebot simulation and not related with the create 3.
I suggest you to re-open the ticket in this repository: https://github.com/turtlebot/turtlebot4_simulator

closed 2024-10-28 by alsora